VOLTTRON/volttron #395: Have WeatherAgent publish data on startup

jhaack, 2016-03-11:
The WeatherAgent has a long poll time and so it's not obvious it is working when it first starts up. By doing an initial publish it would be obvious everything is working.

craig8, 2016-10-26:
@jhaack I know we have an update to this agent that was created.   I don't think that this update addresses this, nor do I know if this is still an issue.

shwethanidd, 2020-02-17:
@jklarson can you comment

jklarson, 2020-02-17:
I suspect this is overcome by events as there is now a base weather service and multiple implemented service agents. I can check it out in the near future.

jhaack, 2020-02-17:
The issue with this was that it was hard to tell if the original weather agent was working if it had a long interval since you would have to wait that time period before getting your first publish. Does the new weather agent publish on startup? If so then this issue can be closed.

jklarson, 2020-02-17:
The new service ages do publish on startup, this issue can be closed.

closed 2020-02-18 by schandrika