VOLTTRON/volttron #398: Control user actions toward agents in volttron central

jchap-pnnl, 2016-03-12:
An agent's properties should indicate whether a user may or may not start or stop it in Volttron Central, and the UI should behave accordingly.

[ ] Include flags in agent objects returned by service for "can start," "can stop," and "can restart."
[ ] Use flags to enable, disable, and hide Start/Stop/Restart buttons.

jchap-pnnl, 2016-03-12:
VC used to make a call to status_agents to determine whether an action was pending. Based on that, combined with the process_id and the status return_code, the UI indicates whether the agent is starting, stopped, was never stopped, etc. 

Should we include something like that (an action_pending, a status return_code) with the agent objects returned by list_platforms?

jchap-pnnl, 2016-03-12:
Or do we just want to keep making the separate status_agents call to find out if the agent is running or not?

craig8, 2016-05-05:
Every agent now has a permissions object.  This object is populated by VC on the serverside.  There are properties can_start, can_stop, can_restart.  and we can add more or use combinations of them to produce the required output.

closed 2016-05-07 by craig8