VOLTTRON/volttron #399: figure out how to autodoc our structure of agents for readthedocs

craig8, 2016-03-13:
The issue is that our instantiation of our agent classes are inside a function wrapper.

We can add paths for each of the agent directories i.e. services/core/ActuatorAgent and then add an entry for the actual agent package that we want to document.  That will get the function documentation, but will not get the internal ActuatorAgent class itself (which is where all the good document is at).

The following mentions a possible work around that may work, but may require reworking how our agents start. 

http://stackoverflow.com/questions/12039633/how-to-autodoc-a-function-inside-a-method-in-sphinx

craig8, 2016-03-13:
To add a path that is to be available for autodoc you can add it to the docs/source/conf.py file like the following

''''
#sys.path.insert(0, os.path.abspath('.'))
sys.path.insert(0, os.path.abspath('../../services/core/ActuatorAgent'))
''''
Then in one of the rst files you can add the following to get the module available for sphinx to document.

''''
.. automodule:: actuator.scheduler
    :memebers:
''''

kmonson, 2016-03-16:
The closure style of instantiating agents is a holdover from VOLTTRON 2. It's not needed anymore now that we can use the same methods to register callbacks in the "starting" method. 

The alternative is to put the documentation in the closure function itself, but that leads to confusing auto-generated docs.

I propose that we remove the closures from our agent code to facilitate auto creation of documentation. Everything can (and probably should) be passed in through __init__ as needed. Configuration handling can still be done with a function that instantiates the Agent class.

kmonson, 2016-05-03:
Key agents have been updated and we have a path forward.

closed 2016-05-03 by kmonson