VOLTTRON/volttron #409: Overwrite auth entries does not work as expected

craig8, 2016-03-18:
This may have been seen to in a branch

```
# Calling this twice with the same publickey produces a AuthFileEntryAlreadyExists even though the default for add is to overwrite the entry.

        authfile = AuthFile()
        authentry = AuthEntry(
            credentials="CURVE:{}".format(vcpublickey)
        )

        authfile.add(authentry)
```

craig8, 2016-03-19:
@zbeech I went ahead and fixed this, but please look over the changes that I have made and make sure they make sense to you.

zbeech, 2016-03-22:
@craig8 Your fixes look good to me. I added a few more auth-file tests and I found a couple more bugs. I fixed these in feature/auth-file-fix.

commit 3b7121a3 (2016-03-19): Fixes: #409 Auth now allows overwriting of credentials.

Conflicts:
	volttrontesting/fixtures/volttron_platform_fixtures.py
commit 880c13f1 (2016-03-19): Fixes: #409 Auth now allows overwriting of credentials.

Conflicts:
	volttrontesting/fixtures/volttron_platform_fixtures.py
commit c5ceeb20 (2016-03-19): Merge pull request #410 from VOLTTRON/feature/new_fix_409

Fixes: #409 Auth now allows overwriting of credentials.
commit 7ac8e71d (2016-03-24): fixed AuthFile bugs, fixes #409
closed 2016-05-07 by craig8