VOLTTRON/volttron #419: Auth regular expression as address doesn't work

craig8, 2016-03-24:
@zbeech please look into this.  I may have done something incorrect.  I would like our default setup for auth to be as follows

CURVE:.* 
address="127.*"

So when a volttron starts up this will be it's default state.  That way local connections are capable out of the box.

Do you see any issues with that?

zbeech, 2016-03-24:
Did you try surrounding the regular expressions with a forward slash?
"credentials": "/CURVE:.*/"
"address": "/127.*/"

craig8, 2016-03-24:
I did.  I actually used what was out on the wiki which is the /127.0.0.*/

craig8, 2016-03-24:
@zbeech I know this is off topic but could you move your gevent_timeout function to a util module under volttrontesting/util/utils.py 

.

closed 2016-05-05 by craig8