VOLTTRON/volttron #420: Auth entry removed if all items not entered

craig8, 2016-03-25:
If I manually add the credentials of CURVE:.* in the auth.json file and then use AuthEntry to add an entry the original CURVE:.* is no longer in the file.

zbeech, 2016-03-25:
I haven't been able to reproduce this error. Here is what I tried:

Manually created /tmp/auth.json:
````JSON
{
  "allow":[
    {
      "comments": "first",
      "credentials":"/CURVE:.*/"
    }
  ],
  "roles":{},
  "groups":{}
}
````

Ran the following script:
````python
from volttron.platform.auth import AuthEntry, AuthFile
auth_file = AuthFile('/tmp/auth.json')
entry = AuthEntry(credentials='NULL', comments='second')
auth_file.add(entry)
````

Now /tmp/auth.json has both entries (the first entry has all its default values set):
````JSON
{
  "allow":[
    {
      "domain":null,
      "user_id":null,
      "roles":[],
      "enabled":true,
      "capabilities":[],
      "comments":"first",
      "groups":[],
      "address":null,
      "credentials":"/CURVE:.*/"
    },
    {
      "domain":null,
      "user_id":null,
      "roles":[],
      "enabled":true,
      "capabilities":[],
      "comments":"second",
      "groups":[],
      "address":null,
      "credentials":"NULL"
    }
  ],
  "roles":{},
  "groups":{}
}
````

craig8, 2016-03-25:
@zbeech 

Please try
{
  "allow":[
    {
      "credentials":"/CURVE:.*/"
    }
}

As I said I ONLY entered the credential.

PR #426 (2016-03-31): Feature/auth file fix
- fixes #420 
- fixes auth-file tests

---
This change is [[image]](https://reviewable.io/reviews/volttron/volttron/426)
commit e4176945 (2016-03-31): fixed bug in AuthFile.add, fixes #420
closed 2016-05-07 by craig8