VOLTTRON/volttron #443: Ability to add devices using VC UI

jchap-pnnl, 2016-04-01:
Implement features in the Volttron Central web interface for adding devices. Steps include scanning for BACnet devices, configuring devices, and editing the registry configuration files.

jchap-pnnl, 2016-04-01:
These are the mockups for the design that will be implemented:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

craig8, 2017-01-03:
We have merged into develop now woo hoo!

closed 2017-01-03 by craig8