VOLTTRON/volttron #457: Multiple bacnet devices on same network.

BalazsHoranyi, 2016-04-17:
master-driver is not working properly when there are multiple bacnet devices under the same I.P address. The grab_bacnet_config.py successfully creates the proper csv files when pointed at the device id.  But when scrapping devices using the master-driver, only a few objects are found and I get a whole bunch of unknown object errors.  Am I missing something or is there a way to implement this?

kmonson, 2016-04-26:
Sorry for the late reply. 

When you run bacnet_scan.py what is the address reported by the utility if you specify a device id range limited to a specific device you are trying to scrape?

You can do this with the command:

python bacnet_scan.py --range 540011 540011

This will give you back only one response from device id 540011 in this form:

Device Address        = <RemoteStation 1002:11>
Device Id             = 540011
maxAPDULengthAccepted = 480
segmentationSupported = segmentedBoth
vendorID              = 5

I think what is going on is that you are giving the wrong address to the driver. For devices behind an MS/TP router you will use a Remote Station address and not the IP address of the router. You can get all of these addresses with the bacnet_scan utility.

You should be able to work with as many devices as you want behind the router as you want.

For the above device my device configuration would look something like this:

{
    "driver_config": {"device_address": "1002:11",
                      "device_id": 540011
                      },
    "campus": "pnnl",
    "building": "isb2",
    "unit": "vav",
    "driver_type": "bacnet",
    "registry_config":"/home/volttron-user/configs/vav.csv"
}

BalazsHoranyi, 2016-04-27:
Thank you for the detailed response. That worked for 90% of my devices. However when I send a who is request to some of my devices behind a router the response looks like this
;Device   MAC (hex)            SNET  SADR (hex)           APDU
;-------- -------------------- ----- -------------------- ----
  12200   0A:07:14:F0:BA:C0    10004 00:40:AE:00:FA:6A    480

How would I go about addressing this in the <network>:<address> format?

kmonson, 2016-04-27:
I'm very sorry. In my previous comment a very important bit was cut out because of markdown formatting.

The response to bacnet_scan should read:

    Device Address = <RemoteStation 1002:11>
    Device Id = 540011
    maxAPDULengthAccepted = 480
    segmentationSupported = segmentedBoth
    vendorID = 5

The "Device Address" line has two parts. The first part is the address type. In this case "RemoteStation" is the address type. Another common type is "Address" which you will see for IP addresses. We do not use the address type in the driver configuration. The library we use detects it automatically from the format of the address.

The second part is the address itself. This must be copied verbatim into the driver configuration as a string. It can take the form of an IP address, a two part remote station address as in the example above, or a hex address in the form 00:40:AE:00:FA:6A. (I think this is the address you need to use based on your WhoIs response. I might have the length wrong. Point bacnet_scan at the device just to be sure.) 

The form of the hex address might also omit the colons. I don't have a device to test against right now so I'm working from memory. In either case the what ever is in the second part of the "Device Address" must be copied verbatim, put into the configuration file, and add quotes.

The "<" and ">" are printed in the actual output but are not a part of the address.

You would use the above hex address like this:

{
"driver_config": {"device_address": "00:40:AE:00:FA:6A",
"device_id": 540011
},
"campus": "pnnl",
"building": "isb2",
"unit": "vav",
"driver_type": "bacnet",
"registry_config":"/home/volttron-user/configs/vav.csv"
}

If there is a problem using the address given by bacnet_scan for the device let me know.

BalazsHoranyi, 2016-05-02:
Works Great! Thanks so much.

kmonson, 2016-05-02:
Awesome!

closed 2016-05-02 by kmonson