VOLTTRON/volttron #463: Changes to auth.json not automatically picked up by platform

jhaack, 2016-04-20:
Manual changes to auth.json are no longer automatically picked up by the platform. Is this by design?

zbeech, 2016-04-20:
This is not by design. I'll look into it.

zbeech, 2016-04-20:
@jhaack Will you please include steps to reproduce this error? It works in my tests. This is what I did:

1. In one terminal start volttron in verbose mode: `volttron -vv`
2. Make a change to `~/.volttron/auth.json` (I did not set VOLTTRON_HOME to something else.)
3. VOLTTRON prints out a log entry: `volttron.platform.auth INFO: loading auth file`

If you don't see the log entry at step 3, then there is indeed a problem.

jhaack, 2016-04-20:
Unable to replicate. Something to keep an eye out for.

jhaack, 2016-10-26:
auth.json should not be modified manually now that there is the auth command.

closed 2016-10-26 by zbeech