VOLTTRON/volttron #464: Revamp how and what is logged at what level for ALL of VOLTTRON.

fstshrk, 2016-04-21:
The current logging for VOLTTRON is not granular enough and does not convey actionable information succinctly.  We decided that we will use three debug levels:

DEBUG -- for truly debug messages
INFO -- for things that convey information about work being done and state of the system.
WARNING --- For things that CAN NOT be ignored and are considered BADDDDDDD.

fstshrk, 2016-04-26:
More information on logging.

DEBUG level should be used for the developer of an agent (or platform). When DEBUG level is enabled for that agent or platform component, the logs will reveal the detailed inner flow of the code to the developer or a support engineer.  DEBUG level should be disabled by default since it is by definition "too much information".

INFO level should be enabled by default and will contain messages for actions such as "state changes," major events or actions. If INFO level logs are printing more than 10-20 logs per minute, then we should look at whether that log message is truly INFO or better classified as DEBUG level. The INFO messages should contain enough detail in the message for a user to be able to understand what is going on.

WARNING (or ERROR) level messages are printed when there is an error encountered by the application or the platform component. However, if the ERROR is such that many instances of the error may occur in a short time span, a type of rate limiting on the message is highly recommended. A WARNING or ERROR message should contain enough information for a developer to be able to at least discover a probable path through the code that may have caused the error.

jhaack, 2016-04-26:
We also need the ability to change the logging level for individual agents. If I'm debugging the historian I don't want to see all the DEBUG info for the rest of the platform.

craig8, 2016-10-26:
We are getting better at this.  There isn't a way to do this on the fly, but through the configstore we will be able to do this.

closed 2017-05-26 by craig8