VOLTTRON/volttron #481: VC TODO

craig8, 2016-04-29:
- [ ] Add lookup of identity through agent uuid.

jchap-pnnl, 2016-04-29:
To dynamically add devices to the tree, the data returned by the devices query will be fit into the following structure:

"platforms":` {
- "4687fedc-65ba-43fe-21dc-098765bafedc": {
  - "uuid": "4687fedc-65ba-43fe-21dc-098765bafedc",
  - "name": "CAMPUS1",
  - "status": "GOOD",
  - "children": ["buildings"],
  - "path": ["platforms", "4687fedc-65ba-43fe-21dc-098765bafedc"],
  - "buildings": {             
      - "path": ["platforms", "4687fedc-65ba-43fe-21dc-098765bafedc", "buildings"],
      - "name": "Buildings",
      - "status": "GOOD",
      - "children": ["1111fedc-65ba-43fe-21dc-098765bafede"],
      - "1111fedc-65ba-43fe-21dc-098765bafede": {
        - "uuid": "1111fedc-65ba-43fe-21dc-098765bafede",
        - "name": "BLG1",
        - "status": "GOOD",
        - "children": ["devices"],
        - "path": ["platforms", "4687fedc-65ba-43fe-21dc-098765bafedc", "buildings", "1111fedc-65ba-43fe-21dc-098765bafede"],
        - "devices": {       
          - "path": ["platforms", "4687fedc-65ba-43fe-21dc-098765bafedc", "buildings", "1111fedc-65ba-43fe-21dc-098765bafede", "devices"],
          - "name": "Devices",
          - "status": "GOOD",
          - "children": ["1231fedc-65ba-43fe-21dc-098765bafedl"],
          - "1231fedc-65ba-43fe-21dc-098765bafedl": {
            - "uuid": "1231fedc-65ba-43fe-21dc-098765bafedl",
            - "name": "RTU1",
            - "status": "GOOD",
            - "path": ["platforms", "4687fedc-65ba-43fe-21dc-098765bafedc", "buildings", "1111fedc-65ba-43fe-21dc-098765bafede", "devices", "1231fedc-65ba-43fe-21dc-098765bafedl"],

jchap-pnnl, 2016-05-02:
Can we change the object returned by the get_devices method to:

result: {
  - path: "PNNL/2400_STEVENS/HEATER1",
  - health: {
    - "status": "GOOD",
    - "last_updated": "2016-05-02T22:28:34.079391+00:00",
    - "context": null }
  - "last_published_utc": "2016-05-02T22:28:31.851270+00:00",
  - "points": [ "HPWH_Phy0_PowerState", "Heartbeat", "temperature", "ERWH_Phy0_ValveState"] } }

That way the front end can get the path by result.path rather than have to do a loop to get the lone key attribute? Also health.status would be consistent with list_platforms and list_agents.

jchap-pnnl, 2016-05-02:
I deleted the previous comment, realizing it wouldn't be a loop for a single key attribute. Rather, there will be N keys for N device objects. That works then. But health.status would still be better.

closed 2016-05-10 by craig8