VOLTTRON/volttron #487: Non-vc instance can't open client

jchap-pnnl, 2016-05-03:
A volttron instance running only a platform agent hangs, repeatedly outputting 

CURVE I: cannot open client HELLO -- wrong server key?
CURVE I: cannot open client HELLO -- wrong server key?
CURVE I: cannot open client HELLO -- wrong server key?

craig8, 2016-05-03:
@jchap-pnnl  I have been unable to reproduce this.  I have 3 instances registered and have started stopped them and reinstalled in several orders.  I haven't been able to get this issue.  

I did also use the volttron-cfg to build the instances.

jchap-pnnl, 2016-05-04:
I cleared out VOLTTRON_HOME for the volttron central instance, and rebuild/restarted everything, and the problem went away.

craig8, 2016-05-05:
I am going to move this to a later version and we can attempt to reproduce this as we go along.

craig8, 2016-10-26:
Unfortunately this a zmq issue.