VOLTTRON/volttron #497: web should be able to support https

craig8, 2016-05-07:
- [ ] Web services should be able to be published through https but can use http

ocschwar, 2017-01-27:
Is this milestone going to be addressed for 4.0?

ocschwar, 2017-01-27:
I hardcoded this diff and tested it, and HTTPS works: 

`+++ b/volttron/platform/web.py
@@ -686,7 +686,10 @@ class MasterWebService(Agent):
             os.makedirs(logdir)
 
         self.appContainer = WebApplicationWrapper(self, hostname, port)
-        svr = WSGIServer((hostname, port), self.appContainer)
+        svr = WSGIServer((hostname, port), self.appContainer,
+                         keyfile="/home/volttron/letsencrypt.key",
+                         certfile="/home/volttron/letsencrypt.crt"
+        )
         self._server_greenlet = gevent.spawn(svr.serve_forever)
`

All I need to know is how to make this a Voltron-wide configuration variable so that SSL becomes an option.

craig8, 2017-01-27:
Yeah I'll do it over the weekend...if it works on our Dev then it's in for 4.1 anyways

Sent from my iPad

> On Jan 27, 2017, at 1:38 PM, ocschwar <notifications@github.com> wrote:
> 
> )

ocschwar, 2017-01-27:
Whoops. Correction. It works only so long as you configure VolttronCentralPlatform to use 
https in the vc_address. 

I did the following and failed to get platform.agent to work correctly: 

> (volttron)volttron@pnpserver:~/volttron/services/core/VolttronCentralPlatform$ volttron-ctl config get platform.agent config
>{
> "volttron_central_address": "https://localhost:8080/"
>}

And am continuing to get this: 

Jan 27 22:07:20 pnpserver volttron[6034]: 2017-01-27 22:07:20,659 (vcplatformagent-3.6.0 7247) volttron.platform.web INFO: Connecting to: http://45.79.138.95:8080/discovery/
Jan 27 22:07:20 pnpserver volttron[6034]: 2017-01-27 22:07:20,660 (vcplatformagent-3.6.0 7247) requests.packages.urllib3.connectionpool INFO: Starting new HTTP connection (1): 45.79.138.95
Jan 27 22:07:20 pnpserver volttron[6034]: Traceback (most recent call last):
Jan 27 22:07:20 pnpserver volttron[6034]: File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 534, in run
Jan 27 22:07:20 pnpserver volttron[6034]: result = self._run(*self.args, **self.kwargs)
Jan 27 22:07:20 pnpserver volttron[6034]: File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/gevent/baseserver.py", line 25, in _handle_and_close_when_done
Jan 27 22:07:20 pnpserver volttron[6034]: return handle(*args_tuple)
Jan 27 22:07:20 pnpserver volttron[6034]: File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/gevent/server.py", line 126, in wrap_socket_and_handle

ocschwar, 2017-01-28:
Adding this --volttron-central-address=https://pnpserver.cse.fraunhofer.org:8080 fails to solve the problem because the WSGIServer serves queries from outside the host but not local queries.

ocschwar, 2017-02-06:
Full implementation can be viewed as of this commit: 

https://github.com/FraunhoferCSE/volttron/commit/d06b5c0ef6e750453995275807621606d51d6ef4#diff-7461bf255e1e2c90021b23c63c57f32e

Changes to main.py and web.py add --web-certfile and --web-keyfile to enable loading certs for HTTPS for Volttron Central. Tested. Works.

shwethanidd, 2020-02-20:
Already fixed in VOLTTRON 6.0

closed 2020-02-20 by shwethanidd