VOLTTRON/volttron #513: Update VC's old chart area to use new chart features

jchap-pnnl, 2016-05-10:


jchap-pnnl, 2016-06-24:
Previous commit (5093af3) also included removing platform UUID from getting chart topics and added getting VC platform to checking for historian running. Using the platform UUID to figure out if it was the VC platform and then check its historian caused problems for when the vc platform wasn't registered (issue #592).

The handling of the platform UUID will have to be added back in if/when we get topics and add charts for remote platforms, too. Currently, all charts and topics -- even those for remote platforms -- are only reported to VOLTTRON Central through the VC platform.

jhaack, 2016-10-27:
@jchap-pnnl Complete?

jchap-pnnl, 2016-10-27:
Yes, it's complete.

closed 2016-10-27 by jhaack