VOLTTRON/volttron #521: sudo historian scripts don't make sure user is priveleged

craig8, 2016-05-12:
When running the sudo install mongodb scripts the user should be prompted to become sudo or should be denied access if they aren't sudo.

schandrika, 2016-06-28:
Fixed the script to check for sudo access and also exit on error.

schandrika, 2016-06-28:
Tested on centos 7 and ubuntu 14.4 and script works as expected. On ubuntu script seem to exit on its own on the first command that returns non zero exit code so the last information message of "script could not complete successfully because of the above error" does not get printed on Ubuntu. If that is okay, I think we can close this issue

craig8, 2016-06-28:
Shouldn't it exit successfully?

schandrika, 2016-06-28:
I don’t understand. May be I didn’t word it correctly

If you have script with
Command_1
Command_2
Command_3

On centos id command_1 gives non zero exit code script proceeds to  command_2, so added logic to check exit code of command 1 before running command_2. In Ubuntu, if command_1 fails I see the error message on console and it doesn’t even call my method for checking exit codes.  Not sure if it is a Ubuntu thing or its some kind of admin settings.

craig8, 2017-06-04:
Shouldn't be too difficult to build a test that travis can launch to install the databases.