VOLTTRON/volttron #532: Handle 401 unauthorized when agent uploade

craig8, 2016-05-15:
This or we should remove the box from install if the user permissions doesn't have can_install permissions.

jchap-pnnl, 2016-05-23:
The handle401() function checks the error code in the response. But in some cases, the responses coming back don't have 401 in the error code -- the codes are undefined. But the error message is "Server returned 401 status." 

Can we add 401 to the error code in the response, so we don't have to parse the message looking for "401" in the string?

craig8, 2016-05-23:
This response is different as it is not a application/json response. My thought is that you would hook into the fact that the http response was a 401 rather than parsing it.

Now our response is more specific, however this does not get as close to what you would like.
<h1>Invalid username/password specified.</h1>
<h2>CODE:-32001</h2>

closed 2017-06-04 by craig8