VOLTTRON/volttron #533: Issue with saving charts

craig8, 2016-05-15:
I believe that this issue has to do with the saving/loading of charts to the store.  To reproduce

1. Create a dashboard chart
1. Add a couple other charts that are not on the dashboard
1. Log out
1. Log in
1. Click on an leaf in the tree

The following image is how mine showed after that even though I only had clicked on one thing the other two charts that I had (not including the one that was in the dashboard) shows up. 

[image]

jchap-pnnl, 2016-05-26:
The steps outlined above don't reproduce the bug anymore. The issue was posted before the synchronization between the tree and the charts was finished. I recommend removing this issue.

craig8, 2016-05-26:
Yup couldn't reproduce

closed 2016-05-26 by craig8