VOLTTRON/volttron #541: Drop support for PLAIN authentication

zbeech, 2016-05-16:
According to http://rfc.zeromq.org/spec:27:

"PLAIN makes no attempt at security or confidentiality. It is intended for use on internal networks where security requirements are low."

Although we have support for PLAIN authentication in volttron/platform/vip/socket.py, it is not used by the rest of the platform. If we drop PLAIN authentication we will be left with NULL (no encryption) and CURVE (strong encryption and authentication). I don't see a good use case for the PLAIN mechanism.

craig8, 2016-10-14:
I don't really think we actually are able to use PLAIN...even if we had it set up.

In addition...I believe we are going to be getting rid of --developer-mode shortly.  I would like to look into what it would take to use certs or gssapi as the method of entry however.

shwethanidd, 2020-02-17:
We no longer use PLAIN authentication for zmq message bus.

closed 2020-02-17 by shwethanidd