VOLTTRON/volttron #557: Chart not shown up with an item checked.

craig8, 2016-05-19:
If a platfrom historian isn't available, but the user still clicks on an item in the tree, the item is till checked.  The error that shows up indicates that there isn't a platform historian.  Then after adding a platform historian the user does not have a chart for that selection.

Expected behavior in my mind is that if an error such as the clicking of a check box is thrown then the checkbox shouldn't be checked (eg have an x in it)

jchap-pnnl, 2016-05-19:
How about if we just uncheck the box if the chart can't be loaded?

craig8, 2016-05-19:
sure, but generally that would happen if there was an error...and the status should pop up as well.

closed 2017-06-04 by craig8