VOLTTRON/volttron #562: Changing the polling frequeny doens't seem to have any effect

craig8, 2016-05-20:
The graphs still updated every once in a while however data is being retrieved at a higher frequency.

jchap-pnnl, 2016-05-20:
Actually, the graphs update as fast as the data is retrieved. The problem is, the data doesn't change any quicker than about once every 30 seconds. If that's a setting that's determined by the devices and can't be changed, it may not make sense to make a chart's Refresh Rate something users can change. Or we should at least give them discreet options for making it slower -- e.g., "Choose from the following: (a) 1x / 30 secs, (b) 1x / 60 secs, (d) 1x / 5 mins, or (e) 1x / 10 mins," if slower updates would even make sense.

craig8, 2016-05-25:
Granted for performance statistics I believe I send those across the wire ever 15 seconds.  So with that it would make sense to be able to change it.