VOLTTRON/volttron #565: Enable aggregation of data points

jhaack, 2016-05-20:
This will require changes to the drivers and historians:

- Drivers include new column which indicates how to aggregate this point (sum, avg, etc.)
- Future work: aggregation type would be automatic based on data model (Temps use avg)
- Create aggregator agents which store aggregated data for selected points. Agents can then query historian and specify aggregated data
- Aggregator agents are configured by:
  - Points to aggregate
  - How many missing data points to allow for missing points
  - time period (hourly, daily, etc)
 - Future work: aggregate historical data

craig8, 2016-10-27:
So @schandrika it looks like not all of this is merged can you take a look at that?

schandrika, 2016-10-27:
The first two points are not implemented but those are not in scope for the first phase of this implementation.

jhaack, 2016-10-27:
#1001 #1002 Capture the points not implemented. Closing issues.

closed 2016-10-27 by jhaack