VOLTTRON/volttron #570: handle subdevices in vc panel tree

jchap-pnnl, 2016-05-21:
BUILDING1 isn't loading in vc's panel tree. It has subdevices, which haven't been fully implemented and tested in the tree yet.

craig8, 2016-05-21:
It did not work on my test.

[image]

craig8, 2016-05-21:
Note my above test was configured like the following
```
{
    "driver_config": {},
    "campus": "PNNL",
    "building": "BUILDING1",
    "unit": "HEATER2",
    "path": "bogus",
    "driver_type": "fakedriver",
    "registry_config":"/home/vdev/git/volttron/services/core/MasterDriverAgent/master_driver/fake.csv",
    "interval": 30,
    "timezone": "US/Pacific",
    "heart_beat_point": "Heartbeat"
}
```

There were 6 other heaters that were connected as well that did not have any sub devices.

craig8, 2016-05-21:
A call to get devices from the platform gives the following response

```
{
	"jsonrpc": "2.0",
	"id": "72581-4",
	"result": {
		"PNNL/BUILDING11/HEATER7": {
			"last_published_utc": "2016-05-21T02:54:43.977084+00:00",
			"points": ["SampleLong3", "OutsideAirTemperature3", "OutsideAirTemperature2", "OutsideAirTemperature1", "SampleBool2", "EKG", "SampleWritableShort1", "SampleWritableShort2", "SampleWritableShort3", "SampleWritableBool1", "SampleWritableFloat1", "SampleWritableBool3", "SampleWritableBool2", "SampleWritableFloat2", "SampleLong2", "SampleBool3", "Heartbeat", "SampleBool1", "SampleLong1", "SampleWritableFloat3"],
			"health": {
				"status": "GOOD",
				"last_updated": "2016-05-21T02:58:24.071673+00:00",
				"context": null
			}
		},
		"PNNL/BUILDING10/HEATER6": {
			"last_published_utc": "2016-05-21T02:54:43.963113+00:00",
			"points": ["SampleLong3", "OutsideAirTemperature3", "OutsideAirTemperature2", "OutsideAirTemperature1", "SampleBool2", "EKG", "SampleWritableShort1", "SampleWritableShort2", "SampleWritableShort3", "SampleWritableBool1", "SampleWritableFloat1", "SampleWritableBool3", "SampleWritableBool2", "SampleWritableFloat2", "SampleLong2", "SampleBool3", "Heartbeat", "SampleBool1", "SampleLong1", "SampleWritableFloat3"],
			"health": {
				"status": "GOOD",
				"last_updated": "2016-05-21T02:58:24.073018+00:00",
				"context": null
			}
		},
		"PNNL/BUILDING4/HEATER1": {
			"last_published_utc": "2016-05-21T02:54:43.985580+00:00",
			"points": ["SampleLong3", "OutsideAirTemperature3", "OutsideAirTemperature2", "OutsideAirTemperature1", "SampleBool2", "EKG", "SampleWritableShort1", "SampleWritableShort2", "SampleWritableShort3", "SampleWritableBool1", "SampleWritableFloat1", "SampleWritableBool3", "SampleWritableBool2", "SampleWritableFloat2", "SampleLong2", "SampleBool3", "Heartbeat", "SampleBool1", "SampleLong1", "SampleWritableFloat3"],
			"health": {
				"status": "GOOD",
				"last_updated": "2016-05-21T02:58:24.073937+00:00",
				"context": null
			}
		},
		"PNNL/BUILDING1/HEATER2/bogus": {
			"last_published_utc": "2016-05-21T02:54:43.954339+00:00",
			"points": ["SampleLong3", "OutsideAirTemperature3", "OutsideAirTemperature2", "OutsideAirTemperature1", "SampleBool2", "EKG", "SampleWritableShort1", "SampleWritableShort2", "SampleWritableShort3", "SampleWritableBool1", "SampleWritableFloat1", "SampleWritableBool3", "SampleWritableBool2", "SampleWritableFloat2", "SampleLong2", "SampleBool3", "Heartbeat", "SampleBool1", "SampleLong1", "SampleWritableFloat3"],
			"health": {
				"status": "GOOD",
				"last_updated": "2016-05-21T02:58:24.074781+00:00",
				"context": null
			}
		},
		"PNNL/BUILDING6/HEATER4": {
			"last_published_utc": "2016-05-21T02:54:43.968190+00:00",
			"points": ["SampleLong3", "OutsideAirTemperature3", "OutsideAirTemperature2", "OutsideAirTemperature1", "SampleBool2", "EKG", "SampleWritableShort1", "SampleWritableShort2", "SampleWritableShort3", "SampleWritableBool1", "SampleWritableFloat1", "SampleWritableBool3", "SampleWritableBool2", "SampleWritableFloat2", "SampleLong2", "SampleBool3", "Heartbeat", "SampleBool1", "SampleLong1", "SampleWritableFloat3"],
			"health": {
				"status": "GOOD",
				"last_updated": "2016-05-21T02:58:24.077082+00:00",
				"context": null
			}
		},
		"PNNL/BUILDING8/HEATER1": {
			"last_published_utc": "2016-05-21T02:54:43.978668+00:00",
			"points": ["SampleLong3", "OutsideAirTemperature3", "OutsideAirTemperature2", "OutsideAirTemperature1", "SampleBool2", "EKG", "SampleWritableShort1", "SampleWritableShort2", "SampleWritableShort3", "SampleWritableBool1", "SampleWritableFloat1", "SampleWritableBool3", "SampleWritableBool2", "SampleWritableFloat2", "SampleLong2", "SampleBool3", "Heartbeat", "SampleBool1", "SampleLong1", "SampleWritableFloat3"],
			"health": {
				"status": "GOOD",
				"last_updated": "2016-05-21T02:58:24.078414+00:00",
				"context": null
			}
		}
	}
}
```

jchap-pnnl, 2016-05-23:
The code assumes that if the list returned by get_devices includes a subdevice, it'll also include a separate object for the the parent device.

craig8, 2016-05-24:
Verified that the fixes do work as defined.

closed 2016-05-24 by craig8