VOLTTRON/volttron #571: Strange issue with vc GUI

craig8, 2016-05-23:
[image]

jchap-pnnl, 2016-05-23:
It's the ZoneTemperature point for three different devices that are all in BUILDING10. If you hover over the chart, you get the breakdown:

[image]

The real issue is, the chart's legend labels probably need to be improved. A quick option is to reverse the order so the device is first and the platform is last, meaning the platform part would be truncated if there wasn't enough room, not the device.

We may be able to come up with a different legend label altogether that doesn't truncate any of the information, but that will take some doing.

jchap-pnnl, 2016-05-23:
Also, you get the whole device path when hovering over the legend label:

[image]

craig8, 2016-05-24:
I see the issue thanks...closing this as a mis diagnosis.

closed 2016-05-24 by craig8