VOLTTRON/volttron #593: Remove saved charts when a platform is deregistered

jchap-pnnl, 2016-05-26:
When a platform is deregistered in vc, its saved charts need to be cleared out. Otherwise, vc will try to load the charts even though the platform isn't in the UI anymore.

jchap-pnnl, 2016-05-26:
Once charts are saved by user, the impact of removing saved charts when a platform is deregistered will be that when an admin removes a platform, it also removes any associated charts for readers. Perhaps that's just par for the course for those who are admins and those who are readers, since admins can always remove platforms and readers won't know why -- those platforms just won't be there the next time they log in. We could have a note that says "the chart doesn't exist because the platform was removed by an administrator." But that sounds like more trouble than we need to go to. Just pointing out the implication here for the record in case it needs to be discussed further.

craig8, 2016-05-27:
That is a good point @jchap-pnnl but as you say admin should be communicating with readers about that as well.  So even on our deployment we have buildings that come and go and then come back, but the dashboard would not reflect them coming back because the admin deleted them or whatever.

jchap-pnnl, 2016-05-28:
Still need to test having more than one platform registered with saved charts: When a platform is deregistered, do the charts associated with the remaining platform survive?

jchap-pnnl, 2016-06-03:
Confirmed case of chart with more than one platform's series. However, this was only verified with charts of platform performance data. Still need to verify the case of charts with device point series from different platforms.

closed 2017-06-04 by craig8