VOLTTRON/volttron #629: Lag in UI response when adding and removing agents

jchap-pnnl, 2016-06-10:
When you click Remove to remove an agent, the agent doesn't go away. Sometimes the start/stop button turns to a disabled start button. If you reload the page, the removed agent might go away. Sometimes it takes two or three page reloads before it does.

The behavior is similar when adding an agent.

I verified this behavior in my environment by adding and removing multiple forwarder and listener agents -- that is, multiple agents with the same name -- but I don't think that's the problem, since they have their own UUIDs.

The problem seems to be a lag in updating the database. The rpc calls -- install and remove_agent -- immediately return, and in their THEN clauses, we proceed with querying the database for updated agent lists and statuses. But those subsequent calls (list_agents and status_agent) return lists that don't reflect the addition or removal of an agent.

If the lag in updating the database is unavoidable -- or even whether it is or isn't -- perhaps we should fix the problem by not relying on the backend to update the UI. When an agent is removed, we could just remove it from the clientside list (if there aren't any errors, of course). When an agent is installed, we get the new agent's uuid in the rpc response, and we can get the agent name by parsing the whl file name. We also need status, but we can assume it's Never Started. We wouldn't know whether the agent can be stopped or removed, however, unless those permission rules can be coded into the front end (anything with vcplatform or volttroncentral in the name can't be stopped or removed. Any other rules?).

A problem with that approach, however, is that the user might try to start the agent before we know that it's in the database.

jchap-pnnl, 2016-06-10:
Maybe the lag is from the time it takes to complete the installation -- that is, copying and removing the files to disk. Can we get the rpc methods to wait until all of that is complete before they respond?

jhaack, 2016-10-26:
@jchap-pnnl  Update?

jchap-pnnl, 2016-10-26:
This issue should stay open. We're planning to address it once we've fully integrated websockets into VC. So the platform will be able to notify the UI when the action on the agent has completed.

craig8, 2017-06-04:
Use websockets to hopefully handle this issue