VOLTTRON/volttron #630: Too many status updates when login or refresh

craig8, 2016-06-11:
I have a couple of platforms that are mis behaving.  I don't think refreshing the page should tell  me that again nor when I login.  The proper place for this is the tree not the status bar I think.

[image]

craig8, 2016-06-28:
@jchap-pnnl where are we on this issue?  I would like to not have this show up on the rc1 branch.

To reproduce you create a vc and vcp (same instance or otherwise) then register or rely on the autoregister.  Then turn the vcp off and hit refresh.

jchap-pnnl, 2016-06-28:
I'll get it removed shortly. I think we could say the larger issue is, only show the status message in response to user actions. Registering or deregistering platform fails, alert the user. Starting, stopping, or removing agent fails, alert the user.

So another place to remove the message would be when the charts fail to refresh because the historian stopped running or something like that. The message can be similarly distracting. But as you've pointed out, it might be helpful to add an indicator to the charts to show that the data isn't getting updated. I'll create a separate issue for that.

jhaack, 2016-10-26:
@jchap-pnnl  Close if fixed

jchap-pnnl, 2016-10-26:
It appears this hasn't been fixed -- at least, not in develop. I'll fix it in the config device branch, unless we need it quicker.

craig8, 2016-10-26:
Config devices will work.  That is going to morph into 4.0.1 branch soon.

closed 2016-12-04 by craig8