VOLTTRON/volttron #638: Constant retries when wrong server key is used in VIP call

jhaack, 2016-06-17:
If an agent on platformA attempts to contact platformB with the wrong server key there is the potential for constant retries with the following error showing up on platformB:

CURVE I: cannot open client HELLO -- wrong server key?

https://github.com/zeromq/libzmq/blob/master/src/curve_server.cpp

This needs to be caught and prevented from constantly retrying.

jhaack, 2016-06-17:
This may be a zmq version difference. Current behavior is the agent dies due to a zmq assertion failure.

jhaack, 2016-10-26:
Still an issue, should be documented

zbeech, 2016-12-12:
See http://stackoverflow.com/questions/38534520/how-can-i-debug-a-connection-issue-within-volttron/38596085#38596085 for background on this issue.

zbeech, 2016-12-12:
I think this is out of our control short of modifying ZMQ.

craig8, 2017-01-03:
Seems like this is something that is not going to be able to be fixed by us :( the >dev/null however will make it not as viewable.

closed 2017-01-03 by craig8