VOLTTRON/volttron #639: DataForward vs. ForwardHistorian

jhaack, 2016-06-17:
We need a ForwardHistorian designed specifically to fill the data forward role. The current one assumed that agents on the remote platform would want to interact with the data as if it were live. This is different than forwarder purely for storage when no agent needs live data.

A redesigned forwarder could bundle the data better and/or utilize database capabilities to do it. It would send to a special receiving agent that can interpret the bundle and store it directly to the database. This would also allow it to give a better data verification response. It would cut down on the number of messages being sent but would need further work if the backup sqlite is still the bottleneck.

shwethanidd, 2020-02-18:
@jhaack have we handled this as part of DataMover agent?

jhaack, 2020-02-19:
Yes, DataMover fills this need

closed 2020-02-19 by jhaack