VOLTTRON/volttron #641: Allow resizing of side panel

jchap-pnnl, 2016-06-17:
Let the user drag the right border of the panel to make it bigger. That way, when the tree is deep and the items are being hidden by the panel's right border, the user doesn't have to scroll to the bottom of the panel to get to the horizontal scroller and then use that to scroll to the right.

jchap-pnnl, 2016-06-17:
The horizontal scroller at the bottom of the panel has gone away. To see items in the tree that are hidden by the right border, the user now _has_ to resize the panel by dragging the border. The horizontal scroller went away because it created weird issues where control items such as the button to expand or collapse the panel and the right border itself to shift away from the right edge of the panel.

jhaack, 2016-10-26:
Re-examine for 4.3