VOLTTRON/volttron #651: charts should indicate if refreshing data or not

jchap-pnnl, 2016-06-28:
Whether by intention (refresh rate turned off) or not (historian or forwarder not running), a chart should indicate whether or not it's refreshing the data. This is probably most important for straight line charts, where you simply can't tell if you're getting new data by seeing if the line changes every few seconds.

For issue #630, we've turned off the status indicator messages for chart refresh, so we don't annoy the user every few seconds when the chart tries and fails again to refresh its data.

Instead, we could put a visual cue on the charts for when data is being fetched -- like circular arrows that spin or dot-dot-dots that blink. Or the cue could be the opposite and instead only appear when there's a problem -- for instance, a red light or X that appears in the corner of the chart, or a red border around the whole chart, when the refresh fails.

jhaack, 2016-10-26:
@jchap-pnnl  Status?

jchap-pnnl, 2016-10-26:
This has been addressed with a green refresh circle that spins next to the chart title when the chart refreshes. The user can tell the chart isn't updating if the green circle doesn't ever appear or if the circle appears but never goes away.

closed 2016-10-26 by jchap-pnnl