VOLTTRON/volttron #725: Add RCP calls to AuthService to provide access to AuthFile

zbeech, 2016-08-30:
Currently AuthFile is called directly in a few places to add auth entries. The AuthService should provide an interface to manipulate AuthFile and this interface should be protected with a "EDIT_AUTHFILE" capability. This will help limit access to AuthFile and it should help with a VOLTTRON_HOME issue.

jhaack, 2016-10-26:
Was this resolved, @zbeech

zbeech, 2016-10-26:
@jhaack This has not been implemented. I'll put it in the VOLTTRON 4.1 queue.

zbeech, 2016-12-12:
I can't remember the use case for this feature.

craig8, 2016-12-12:
We would allow certain agents VC, VCP to modify the auth file on the server through this mechanism.  If the connection was granted "SUPER ADMIN" capabilities or something.  Really it's not practical until we look at containers.

schandrika, 2020-02-20:
@schandrika to test access to auth files in volttron secure mode