VOLTTRON/volttron #738: Add help message when executing volttron-ctl config without arguments

craig8, 2016-09-09:
Allow the user to see what options are available when executing volttrono-ctl config.

````
(volttron)vdev@big-mint-cinn ~/git/volttron $ volttron-ctl config
usage: volttron-ctl command [OPTIONS] ... config [-h] [-c FILE] [--debug]
                                                 [-t SECS]
                                                 [--vip-address ZMQADDR] [-k]
                                                 [--keystore-file FILE]
                                                 [--known-hosts-file FILE]
                                                 ...
volttron-ctl command [OPTIONS] ... config: error: too few arguments
````

The same issue with volttron-ctl auth produces the following

````
(volttron)vdev@big-mint-cinn ~/git/volttron $ volttron-ctl auth
usage: volttron-ctl command [OPTIONS] ...
volttron-ctl: error: argument : invalid choice: 'auth' (choose from 'install', 'tag', 'remove', 'list', 'status', 'clear', 'enable', 'disable', 'start', 'stop', 'restart', 'run', 'config', 'shutdown', 'send', 'keypair', 'add-known-host', 'stats', 'serverkey', 'auth-list', 'auth-add', 'auth-remove', 'auth-update')
````

craig8, 2016-09-09:
@kmonson  please take care of this when you get to the doc strings for the config store.

craig8, 2016-09-09:
As @kmonson points out that I have in comparable items in this list.  I am wondering if we should remove the global configs just for the config arguments.  We need to look into that.

kmonson, 2016-09-09:
The example given is not comparable. "auth" is not a valid sub command.

closed 2016-10-05 by kmonson