VOLTTRON/volttron #746: Installing agent issue

craig8, 2016-09-13:
This has been duplicated a couple of times.  Steps are as follows

1. Allow volttron to run for an indeterminate time period (i haven't seen it do this in under a day)
1. Attempt to install an agent on the platform

````
2016-09-13 11:46:24,150 () volttron.platform.web DEBUG: registered route t is: callable
2016-09-13 11:46:24,407 () volttron.platform.control DEBUG: Received 3176390 bytes of data
2016-09-13 11:46:24,407 () volttron.platform.control DEBUG: done receiving data
2016-09-13 11:46:24,407 () volttron.platform.control DEBUG: Sending done back!
2016-09-13 11:46:24,407 () volttron.platform.control DEBUG: Closing channel on server
2016-09-13 11:46:24,409 () volttron.platform.vip.agent.subsystems.rpc ERROR: unhandled exception in JSON-RPC method 'install_agent': 
Traceback (most recent call last):
  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 168, in method
    return method(*args, **kwargs)
  File "/home/volttron/volttron/volttron/platform/control.py", line 287, in install_agent
    agent_uuid = self._aip.install_agent(path, vip_identity=vip_identity)
  File "/home/volttron/volttron/volttron/platform/aip.py", line 296, in install_agent
    unpack(agent_wheel, dest=agent_path)
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/wheel/tool/__init__.py", line 135, in unpack
    sys.stderr.write("Unpacking to: %s\n" % (destination))
IOError: [Errno 5] Input/output error
````

craig8, 2016-09-13:
This seems to happen quicker when there are more agents installed.

craig8, 2016-09-19:
This issue is probably exacerbated by running on boards that may not have fast enough memory processing power.  there may be a race condition with the last writing of the file and the starting of the installation process.

rlutes, 2016-09-19:
Possibly, problem also occurs on NUC running higher end hardware (i7 processor and 8 GB memory).

craig8, 2016-09-20:
Most likely #756 will fix this for local instances.

craig8, 2016-09-21:
Other work has helped to mitigate the local install issues in merge #759

craig8, 2016-10-20:
Seems that this just won't die
````
2016-10-19 20:13:16,226 () volttron.platform.vip.agent.subsystems.rpc ERROR: unhandled exception in JSON-RPC method 'install_agent_local': 
Traceback (most recent call last):
  File "/home/volttron/volttron/volttron/platform/vip/agent/subsystems/rpc.py", line 168, in method
    return method(*args, **kwargs)
  File "/home/volttron/volttron/volttron/platform/control.py", line 256, in install_agent_local
    add_auth=add_auth)
  File "/home/volttron/volttron/volttron/platform/aip.py", line 305, in install_agent
    unpack(agent_wheel, dest=agent_path)
  File "/home/volttron/volttron/env/local/lib/python2.7/site-packages/wheel/tool/__init__.py", line 134, in unpack
    sys.stderr.write("Unpacking to: %s\n" % (destination))
IOError: [Errno 5] Input/output error
````

craig8, 2016-10-20:
Output from top (note volttron wasn't anywhere in the top 5 processes)

````
op - 20:23:43 up 19 days, 11:08,  4 users,  load average: 0.00, 0.01, 0.05
Tasks: 135 total,   1 running, 134 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.1 us,  0.1 sy,  0.0 ni, 99.8 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem:   2038684 total,   886244 used,  1152440 free,   102544 buffers
KiB Swap:        0 total,        0 used,        0 free.   500660 cached Mem
````

craig8, 2016-10-20:
I suppose a good thing is that restarting the platform does fix this and if your agents are installed they aren't affected!!!

jhaack, 2016-10-20:
Workaround is to redirect output to dev/null. Issue is related to ssh'ing into a box to start VOLTTRON, logging off, then getting back on and trying to install.

kmonson, 2016-10-21:
To repoduce: run volttron in the background (with &), detach it, then close your ssh session, log back in, try to install an agent.

The wheel library writes directly to stderr. If the output is not directed to catch this it produces an IO error when attempting to install an agent.

In practice we can't fix it or it is very difficult as we can't definitively say that when the user does not want to see output from stderr. Also we can't redirect output from stderr and stdout to a logger without first determining if the logger itself is sending its output to stdout. If we didn't allow for "anything you want" style logging configuration it could be done as then we have more control and could know what was going where.

kmonson, 2016-10-21:
We'll improve the docs.

jhaack, 2016-10-26:
Mentioned in multiple places in documentation now

closed 2016-10-26 by jhaack