VOLTTRON/volttron #750: Separate tables for different base topics

jhaack, 2016-09-15:
Make separate tables for: analysis, devices, record, etc. This will require separate topic tables as well for each.

Add a parameter to historian query to direct to the appropriate table. Default table is current data table.

craig8, 2016-09-16:
Do we add a parameter or a configuration setting.  This would need to be in all concrete historians

jhaack, 2016-10-26:
@schandrika Does recent work on historian query performance lessen the need for this?

schandrika, 2016-10-26:
Historian queries are still slower when getting data for more than a couple of weeks. I am not sure about the impact of this though as we now support queries across multiple topics. I think this is still worth investigating and running some quick tests.

shwethanidd, 2020-02-19:
@schandrika please comment on current status.

schandrika, 2020-02-20:
@schandrika to do a performance test on production environment and scope work