VOLTTRON/volttron #757: Logging does not seem to work in control.py

craig8, 2016-09-20:
The new logging that you put in on #756 does not show up when running the platform in -vv.   However I know the method is getting executed correctly because of a handy print statement.

mikeroup, 2016-09-20:
Good catch. That is pretty weird. control.py has the control agent and the code for the command line tool so that could be a factor.

mikeroup, 2016-09-20:
volttron-ctl has its own logging level. Try volttron-ctl -vv install

craig8, 2016-09-20:
So this isn't an issue more than it's logging to a different pipe...i.e. we aren't going to get this data into say volttron.log

mikeroup, 2016-09-20:
Correct. There might be a way to attach it to the volttron logger but I would defer to Kyle about that.

craig8, 2016-09-20:
After discussing I believe that this is not an issue and will not be "fixed".   As @mikeroup reports -vv on the volttron-ctl does indeed give the output that I was thinking should go in the log.  And after conversations with the team I believe that is the behavior that should stay.

closed 2016-09-20 by craig8