VOLTTRON/volttron #791: management-service-demo scripts broken

jhaack, 2016-09-30:
Could be replaced with instructions for volttron-cfg since that is much simpler than the original process this script was meant to demonstrate.

jhaack, 2016-10-05:
Take out old docs and make a simple walkthrough using volttron-cfg

ffreckle, 2016-10-10:
I started down the path of trying to fix this demo before seeing this bug.  I think the scripts are useful and a nice way to test out a configuration.  Do you intend to update the scripts to use volttron-cfg or will the scripts be deprecated and replaced by documentation? I'm willing to take a shot at fixing the scripts but may need some assistance.

jhaack, 2016-10-10:
We're planning on replacing them with documentation since volttron-cfg makes it so much easier now. That being said, new scripts could be used as a way to show automating a deployment without using cfg.

mikeroup, 2016-10-20:
#762

closed 2016-10-27 by craig8