VOLTTRON/volttron #811: Identity not being picked up from configuration file

jhaack, 2016-10-05:
"identity" is not being used in old style configuration files (such as platform.historian). 

This will cause confusion with deployments not done through volttron-cfg.

Either pick identity up out of old configuration files or highlight this breaking change.

kmonson, 2016-10-05:
We will highlight the breaking change. How do we want to do that?

craig8, 2016-10-05:
I think that this is a change that we can do, however I would like to be able to modify the historians so that they can read that first and then pass them to the config store.  This ability should be deprecated and will be removed at the 4.1 release.  That gives people time to fix this before we go on.

craig8, 2016-10-05:
We should document this as a future pain point in the release notes.  and in readthedocs as well.

kmonson, 2016-10-05:
There is a TERRIBLE AND AWFUL hack where we pop the identity given to us by vip_main and supply our own. 

The reason why I hate this is that the VIP IDENTITY given by 

    volttron-ctl status

is no longer ground truth for that agent. As the historian does not currently support the configuration store this is not as terrible as it could be.  If we do put in this hack it must be removed when configuration support is added historians.

jhaack, 2016-10-06:
2016-10-05 21:43:22,044 (sqlhistorianagent-3.5.0 8084) volttron.platform.vip.agent.core INFO: Connected to platform: router: 92118517-0c94-448a-a52c-655c4cd15966 version: 1.0 identity: platform.historian
2016-10-05 21:43:22,044 (sqlhistorianagent-3.5.0 8084) sqlhistorian.historian INFO: Starting historian with identity: platform.historian
2016-10-05 21:43:22,074 (sqlhistorianagent-3.5.0 8084) <stderr> ERROR:     self.core.identity).get(timeout=2)

craig8, 2016-10-06:
That means you have one installed already.  I believe

jhaack, 2016-10-06:
Looks like the issue is that the changes in this pull fix it in the agent, looks like it is being started with the correct identity. However, the IDENTITY file is set in packaging which does not pick up that change.

commit 9a5563d8 (2016-10-06): Fixes #811 (#818)

* Fixes #811

* Added deprecation warnings to historian identity configuration settings.

* Updated example mysql historian configuration.
closed 2016-10-17 by kmonson