VOLTTRON/volttron #828: Standalone Listener not working

jhaack, 2016-10-06:
Standalone listener is now failing due to identity changes. Should be updated since this is used as an example of a non-agent application interacting with with the message bus.

StanaloneListenerAuth can probably be removed. Check if it still serves a purpose.

Error from StandaloneListener:

python standalonelistener.py
tcp://127.0.0.1:22916?serverkey=<key>&publickey=<key>secretkey=<key>
2016-10-06 16:43:17,032 root WARNING: Deprecation warining
Traceback (most recent call last):
  File "standalonelistener.py", line 113, in <module>
    identity='Standalone Listener') 
  File "/home/volttron/git/volttron/volttron/platform/vip/agent/__init__.py", line 97, in __init__
    idenity=identity))
KeyError: 'identity'

kmonson, 2016-10-06:
Heh, it's a typo in a warning log message.

jhaack, 2016-10-07:
Ah, typo in error message hides the issue of an identity with a space in it.

commit 0632e887 (2016-10-10): Fixes #828 (#829)
closed 2016-10-17 by kmonson