VOLTTRON/volttron #831: WeatherAgent does not work properly with country/city instead of zip_code

damienpicard, 2016-10-07:
The code of weatherAgent is not correct to use it with country/city instead of zip_code.
The following lines of the file: https://github.com/VOLTTRON/volttron/blob/master/services/core/WeatherAgent/weather/weatheragent.py
```
    agent_id = get_config('agentid')
    poll_time = get_config('poll_time')
    zip_code = get_config("zip")
    key = get_config('key')
    on_request_only = get_config('on_request_only')

    state = ''
    country = ''
    city = ''
    region = state if state != "" else country
    city = city
```
should be replaced by:
```
agent_id = get_config('agentid')
    poll_time = get_config('poll_time')
    zip_code = get_config("zip")
    state = get_config("state")
    country = get_config("country")
    city = get_config("city")
    key = get_config('key')
    on_request_only = get_config('on_request_only')
```

and also:
```
elif self.region != ""
```
by
```
elif region !=""
```

damienpicard, 2016-10-07:
After adapting the code, the agent runs smoothly.

craig8, 2016-10-09:
Great @damienpicard.  If it wouldn't be too much trouble could you see if the develop branch gives you the same issue.  It is what will be in master next week sometime.  Master is the most stable oldest branch.  See http://volttron.readthedocs.io/en/develop/roadmap/index.html for the roadmap.  Currently master is 3.0 code.

damienpicard, 2016-11-12:
@craig8 the same issue remained in the develop branch. I issued a pull request fixing it.

acedrew, 2017-03-05:
Is this resolved? There seems to be a lot of noise in the issues, love to get them pruned down a bit to current problems

PR #1414 (2017-07-10): Fixes #831
Fixes #831 
Based on https://github.com/VOLTTRON/volttron/pull/1047  addresses @craig8 formatting concerns.

---
This change is [[image]](https://reviewable.io/reviews/volttron/volttron/1414)
commit 615c00a9 (2017-07-10): Merge pull request #1414 from kmonson/feature/weather-fix

Fixes #831
commit d81465d5 (2017-07-10): Fixes #831
closed 2018-03-30 by craig8