VOLTTRON/volttron #833: Connection object is trying to connect many thousands of times

craig8, 2016-10-07:
This could be caused by a peer not being available on the platform or the platform being down or another issue...however this must be fixed before 4.0 goes out!

mikeroup, 2016-10-18:
This should happen for any agent. If volttron-ctl is pointed toward a nonexistent volttron eg
```
volttron-ctl status --vip-address=tcp://127.0.0.x:22914
```
 then the same type of traffic will be generated. This is "expected" because bind and connect order does not matter to zmq.

craig8, 2016-10-18:
Yes because zmq doesn't care which end is connected first.   We do however.  For us we would like the internal Connection object https://github.com/VOLTTRON/volttron/blob/develop/volttron/platform/vip/agent/connection.py
to first detect whether the endpoint is active.  You can test this via telnet from the command line.  You should be able to do a simple tcp connection to see if it is active before attempting to connect.  

This will not fix the #487 but will at least lessen the spaming when I shut down vc for a minute or two.  

Have the reconnection period be around 10 seconds.  That should be enough that we shouldn't have too many connection attempts.

mikeroup, 2016-10-18:
We can use
```
socket.setsockopt(zmq.RECONNECT_IVL, 1000) #measured in milliseconds
```
to tell the socket to only try to reconnect once every second. We might need to pass this to the agent and handle it in Core but that should mitigate the issue.

closed 2016-10-27 by craig8