VOLTTRON/volttron #851: Remove --keystore-file and --known-hosts-file options from volttron-ctl

zbeech, 2016-10-11:
By default `volttron-ctl` uses `$VOLTTRON_HOME/keystore` to get its public key and secret key,  and it uses `$VOLTTRON_HOME/known_hosts` to get the server key (platform's public key). The `--keystore-file` and `--known-hosts-file` options allow other files to be used in place of those defaults.

I added these options when I initially added the `KeyStore` and `KnownHostsStore` classes, but now I do not think this options are necessary. In fact, I think they only make `volttron-ctl` more confusing, especially when running the `volttron-ctl auth keypair` command (see #838). I would like to remove these two options.

craig8, 2016-10-11:
Are you doing this before or after the 4.0 release?

zbeech, 2016-10-11:
@craig8 If I get it done this afternoon are you comfortable merging it before the 4.0 release?

zbeech, 2016-10-11:
Either way, I'll try to get it done soon and you can decide if you want to merge it before the release. I think it should be a straightforward change.

craig8, 2016-10-11:
I think so...not sure anyone is using it other than us at present....I will not it in the release notes however.

craig8, 2016-10-11:
So wait does this mean there is only one keystore per platform?  Or how does one specify a different keystore.  Do they have that option any longer?

zbeech, 2016-10-11:
Correct, there is only one keystore per platform. I don't believe we have ever had the option of using a different keystore when starting up the platform. The existing --kestore-file option has only been for volttron-ctl.

PR #856 (2016-10-12): Removed --keystore-file and --known-hosts-file from volttron-ctl
closes #851, fixes #838

---
This change is [[image]](https://reviewable.io/reviews/volttron/volttron/856)
commit daa88a62 (2016-10-12): removed --keystore-file and --known-hosts-file options from volttron-ctl, closes #851, fixes #838 (#856)
closed 2016-10-26 by craig8