VOLTTRON/volttron #861: what is tcp in VIP address?

jun343, 2016-10-12:
i want to make platform in volttron
but VIP address is error that is tcp has 6 address
so i found that however i can`t find tcp.

plz, how to find the tcp?

jhaack, 2016-10-12:
tcp is the ip address of the machine VOLTTRON is deployed on. The tcp: prefix indicates that you'll be using an IP address instead of an ipc address.

craig8, 2016-10-12:
tcp stands for transmission control protocol whilest ipc stands for interprocess communication.  ipc is used when multiple VOLTTRON instances are deployed on a single server and to communicate between the VOLTTRON process and other agent processes.

to find your tcp address you can use 

````
# The new way
ip addr

# The old way
ifconfig
````

Your output will have a 10.0.5.10 format.

closed 2016-10-20 by craig8