VOLTTRON/volttron #908: Add "volttron-ctl auth remove-known-hosts" command

zbeech, 2016-10-19:
Users need a way to remove hosts from the known-hosts-file. This is especially important when a remote platform's public key changes.

craig8, 2016-10-19:
when do you add an entry to the known-hosts file currently?

zbeech, 2016-10-19:
Currently it is an entry can be added to the known-hosts file in two ways:

1. The `volttron-ctl auth add-known-host` command
2. In `main.py`, when the platform starts it adds its own public key so local agents automatically know what server key to use

craig8, 2016-10-19:
And the purpose for 1 is ?  Will it overwrite the key in the known host file?  Is this for external agents connecting to the local platform?

zbeech, 2016-10-19:
Yes, 1 is for agent (on platform A) connecting to external platform B. The user on platform A can run:

`volttron-ctl auth add-known-host --host $B_ADDR --server-key $B_PUB_KEY`

Then anytime agents on platform A need to connect to platform B they know what server key to use.

craig8, 2016-10-19:
This seems like what I should be using in vcm and vc then.  Rather than adding both to auth.json. 

So there isn't a way to do it through an rpc method for vcp at present?

zbeech, 2016-10-19:
Also, yes, the `add-known-hosts` will overwrite the the server key in the known-host file.

zbeech, 2016-10-19:
No, there isn't currently an RPC method for this.

zbeech, 2016-12-12:
merged into develop

closed 2016-12-12 by zbeech