VOLTTRON/volttron #951: volttron-pkg not able have a proper environment in /tmp

ocschwar, 2016-10-25:
I'm checking some modifications to VolttronCentral, but am unable to repackage them: (develop branch)

`(volttron)volttron@pnpserver:~/volttron$ volttron-pkg package services/core/VolttronCentral 
Traceback (most recent call last):
  File "setup.py", line 79, in <module>
    _temp = __import__(agent_module, globals(), locals(), ['__version__'], -1)
  File "/tmp/tmp2OR9hP/pkg/volttroncentral/agent.py", line 108, in <module>
    from volttron.platform.vip.agent.connection import Connection
ImportError: No module named connection
2016-10-25 15:28:30,496 () volttron.platform.packaging ERROR: Command '['/home/volttron/volttron/env/bin/python2.7', 'setup.py', '--no-user-cfg', '--quiet', 'bdist_wheel']' returned non-zero exit status 1
`

craig8, 2016-10-25:
Please tell me what branch you are working on.  I believe that this is caused because the Connection object  has been moved.

Please pull down the latest develop code and rebuild the volttroncentral agent (version is 3.5.5) currently.

ocschwar, 2016-10-25:
It is the develop branch, pulled an hour ago. And IIUC, volttron-pkg is the command I should be using to rebuild.

craig8, 2016-10-25:
Generally we use modifications to the script located in scripts/core make-listener or upgrade-listener.  We place them in a config folder under the root of the volttron directory (the config directory is in .gitignore).

If you do a pip list make sure you have volttron 4.0.0 in the list.  If not then execute python bootstrap.py from your activated shell.

ocschwar, 2016-10-25:
pip list confirms volttron 4.0.0 .

craig8, 2016-10-25:
did you get any merge conflicts when you last pulled?
Is this a new instance?

There is a possibility that your temp directory doesn't allow executable code.

ocschwar, 2016-10-25:
Here's the branch comparison. The file list doesn't point to anything. 

https://github.com/VOLTTRON/volttron/compare/develop...FraunhoferCSE:fh-base

ocschwar, 2016-10-25:
Runnign strike with volttron-pkg shows that /tmp is allowing writeable files  files to  open: 
All calls appear to ask for 0666.

open("/tmp/tmpHChkG5/pkg/package.json", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
open("/tmp/tmpHChkG5/pkg/volttroncentral/webroot/api-tests.html", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
open("/tmp/tmpHChkG5/pkg/volttroncentral/webroot/index.html", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
open("/tmp/tmpHChkG5/pkg/volttroncentral/webroot/js/app-04d33897.js", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
open("/tmp/tmpHChkG5/pkg/volttroncentral/webroot/js/vendor-3ac3895b.js", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
[...]

craig8, 2016-10-25:
What is strike?  Perhaps I can run it on my side and see if that's the culprit.

ocschwar, 2016-10-25:
strace (autocorrect goofup.) I used it to confirm /tmp was not the problem. And it just started working.

craig8, 2016-10-25:
magically?

craig8, 2016-10-25:
hmm did you deactivate/close your console and reactivate?

ocschwar, 2016-10-25:
Yes. I also reran bootstrap.py. So it's one of those two.

craig8, 2016-10-25:
Thanks...glad you got it working.  I am going to close this issue for you.

closed 2016-10-25 by craig8