VOLTTRON/volttron #961: UI Hangs when trying to open a platform in the tree

jhaack, 2016-10-26:
Run through management demo.
Tree hangs when trying to open the tree for a platform running the master driver.
Could be a problem with the UI or with the fakedriver

craig8, 2016-10-26:
Is the platform the same platform as vc or the other one.

craig8, 2016-10-26:
I am almost positive that this is because we don't have a forwardhistorian forwarding to the vc platform.  

I am going to install one now on the secondary platform and see if that does ideed fix the issue.

craig8, 2016-10-26:
Hmmm I am wondering if this has to do with the no route to host for peer that I have seen in testing.  There seems to be a bit of that going around that I need to track down why.

jchap-pnnl, 2016-10-26:
The problem is the client code gets into an infinite loop because the fake device's key is a single string "fake," but the code expects it to be a three-part path like "campus/building/device."

closed 2017-01-03 by craig8