VOLTTRON/volttron #984: SQLHistorian error using MySQL after update to 4.0.

rlutes, 2016-10-26:
This error appears after updating to VOLTTRON 4.0.  Error is eliminated when pulling back to releases/3.5rc1 branch.

-Traceback (most recent call last):
16544463-  File "/home/building-testbed/volttron4-latest/volttron/platform/agent/base_historian.py", line 680, in _process_loop
16544582-    self.publish_to_historian(to_publish_list)
16544629:  File "/home/building-testbed/.volttron/agents/af82a195-ff2f-4593-bfc2-6dff675a6c9a/sqlhistorianagent-3.6.0/sqlhistorian/historian.py", line 240, in publish_to_historian
16544800-    if self.writer.insert_data(ts, topic_id, value):
16544853-  File "/home/building-testbed/volttron4-latest/volttron/platform/dbutils/basedb.py", line 296, in insert_data
16544964-    (ts, topic_id, jsonapi.dumps(data)))
16545005:  File "/home/building-testbed/volttron4-latest/env/local/lib/python2.7/site-packages/mysql/connector/cursor.py", line 515, in execute
16545140-    self._handle_result(self._connection.cmd_query(stmt))
16545198:  File "/home/building-testbed/volttron4-latest/env/local/lib/python2.7/site-packages/mysql/connector/connection.py", line 490, in cmd_query
16545339-    result = self._handle_result(self._send_cmd(ServerCmd.QUERY, query))
16545412:  File "/home/building-testbed/volttron4-latest/env/local/lib/python2.7/site-packages/mysql/connector/connection.py", line 395, in _handle_result
16545558-    raise errors.get_exception(packet)
16545597-DataError: 1406 (22001): Data too long for column 'value_string' at row 1

craig8, 2016-10-26:
This was not reproducible on the box that it was demonstrated on.  However #988 which is even more of a pain was.

craig8, 2016-10-28:
This is actually this is a problem with the schema.  The value_string in our default schema is TEXT which has a max of 64 KiB worth of data.

After investigation the issue is that the data is larger than that so alter the schema by executing

ALTER TABLE data MODIFY value_string LONGTEXT; 

See below or the docs (http://dev.mysql.com/doc/refman/5.7/en/blob.html) for the following sizes.

      Type | Maximum length
-----------+-------------------------------------
  TINYTEXT |           255 (2<sup> 8</sup>&minus;1) bytes
      TEXT |        65,535 (2<sup>16</sup>&minus;1) bytes = 64 KiB
MEDIUMTEXT |    16,777,215 (2<sup>24</sup>&minus;1) bytes = 16 MiB
  LONGTEXT | 4,294,967,295 (2<sup>32</sup>&minus;1) bytes =  4 GiB

closed 2016-10-28 by craig8