VOLTTRON/volttron #988: SQLHistorian has no writer attribute

craig8, 2016-10-26:
This happens sometimes during agent start up.  My guess is there is a race condition between the writer and reader.

I believe that the process loop in the base historian gets backed up data and attempts to write to the writer before the agents writer is fully initialized.

````
2016-10-26 15:44:11,091 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR: Exception in thread Thread-2:
2016-10-26 15:44:11,091 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR: Traceback (most recent call last):
2016-10-26 15:44:11,091 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR:   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
2016-10-26 15:44:11,091 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR:     self.run()
2016-10-26 15:44:11,091 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR:   File "/usr/lib/python2.7/threading.py", line 754, in run
2016-10-26 15:44:11,091 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR:     self.__target(*self.__args, **self.__kwargs)
2016-10-26 15:44:11,091 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR:   File "/home/building-testbed/volttron4-latest/volttron/platform/agent/base_historian.py", line 639, in _process_loop
2016-10-26 15:44:11,092 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR:     self.record_table_definitions(self.volttron_table_defs)
2016-10-26 15:44:11,092 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR:   File "/home/building-testbed/.volttron/agents/26d7b095-9e5d-43df-bdfa-7c551584efd0/sqlhistorianagent-3.6.0/sqlhistorian/historian.py", line 192, in record_table_definitions
2016-10-26 15:44:11,092 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR:     self.writer.record_table_definitions(self.tables_def,
2016-10-26 15:44:11,092 (sqlhistorianagent-3.6.0 1179) <stderr> ERROR: AttributeError: 'SQLHistorian' object has no attribute 'writer'
````

To reproduce restart agent a number of times.  

To fix, remove the agent backup.sqlite database.  from the agent's data directory.

This is an issue because after this error noting is written to the cache nor to the database!

kmonson, 2016-10-27:
The problem is that the connections in SQLhistorian are setup in __init__. One of the reasons for the historian_setup function is to make sure that the connections are setup before the process loop starts. As the process loop is started in the init method in it's own thread it can reach the record_table_definitions before the connection is setup in SQLhistorian.

If there is a REALLY good reason to setup the connection in __init__ instead of historian_setup we could setup the connections before calling super().__init__, otherwise we need to move it to historian setup.

schandrika, 2016-10-27:
Ready to merge

closed 2016-10-28 by craig8

ocschwar, 2016-10-28:
Is this going to be pushed onto the develop branch?

jhaack, 2016-10-28:
This will be pushed out as an update to master