VOLTTRON/volttron #1003: Intermittent failures of connection_test

jhaack, 2016-10-27:
Investigate intermittent failures of the connection test. One of the four builds seems to fail with this fairly regularly but not consistently.

==================================== ERRORS ====================================
___________ ERROR at setup of test_can_connect_to_control[encrypted] ___________

request = <SubRequest 'setup_control_connection' for <Function 'test_can_connect_to_control[encrypted]'>>
get_volttron_instances = <function get_n_volttron_instances at 0x7fe3a22a8b90>

[1m    @pytest.fixture(scope="module")[0m
[1m    def setup_control_connection(request, get_volttron_instances):[0m
[1m        """ Creates a single instance of VOLTTRON for testing purposes[0m
[1m        """[0m
[1m        global wrapper, control_connection[0m
[1m    [0m
[1m        wrapper = get_volttron_instances(1)[0m
[1m    [0m
[1m        request.addfinalizer(wrapper.shutdown_platform)[0m
[1m    [0m
[1m        assert wrapper[0m
[1m        assert wrapper.is_running()[0m
[1m    [0m
[1m        if get_volttron_instances.param == 'encrypted':[0m
[1m            if wrapper.encrypt:[0m
[1m                wrapper.allow_all_connections()[0m
[1m            # Connect using keys[0m
[1m            ks = KeyStore()[0m
[1m            ks.generate()[0m
[1m    [0m
[1m            control_connection = build_connection(identity="foo",[0m
[1m                                                  address=wrapper.vip_address,[0m
[1m                                                  peer=CONTROL,[0m
[1m                                                  serverkey=wrapper.serverkey,[0m
[1m                                                  publickey=ks.public,[0m
[1m>                                                 secretkey=ks.secret)[0m

[1m[31mvolttrontesting/platform/connection_test.py[0m:34: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[1m[31mvolttron/platform/vip/agent/utils.py[0m:78: in build_connection
[1m    publickey=publickey, secretkey=secretkey, **kwargs)[0m
[1m[31mvolttron/platform/vip/agent/connection.py[0m:134: in __init__
[1m    self.is_connected()[0m
[1m[31mvolttron/platform/vip/agent/connection.py[0m:182: in is_connected
[1m    return self.server.core.connected[0m
[1m[31mvolttron/platform/vip/agent/connection.py[0m:172: in server
[1m    if self.peer not in self._server.vip.peerlist().get(timeout=2):[0m
[1m[31mvolttron/platform/vip/agent/subsystems/peerlist.py[0m:85: in list
[1m    socket.send_vip(b'', b'peerlist', [b'list'], result.ident)[0m
[1m[31mvolttron/platform/vip/socket.py[0m:418: in send_vip
[1m    flags=flags|more, copy=copy, track=track)[0m
[1m[31mvolttron/platform/vip/socket.py[0m:392: in send_multipart
[1m    msg_parts, flags=flags, copy=copy, track=track)[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/zmq/green/core.py[0m:250: in send_multipart
[1m    msg = super(_Socket, self).send_multipart(*args, **kwargs)[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/zmq/sugar/socket.py[0m:366: in send_multipart
[1m    self.send(msg, SNDMORE|flags, copy=copy, track=track)[0m
[1m[31mvolttron/platform/vip/socket.py[0m:384: in send
[1m    frame, flags=flags, copy=copy, track=track)[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/zmq/green/core.py[0m:215: in send
[1m    self._wait_write()[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/zmq/green/core.py[0m:137: in _wait_write
[1m    self.__writable.get(block=True)[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/gevent/event.py[0m:370: in get
[1m    self._wait_core(timeout, ())[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/gevent/event.py[0m:95: in _wait_core
[1m    result = self.hub.switch()[0m
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <Hub at 0x7fe3a2325910 epoll default pending=0 ref=8 fileno=3>

[1m    def switch(self):[0m
[... 202 lines cut ...]
[1m    msg_parts, flags=flags, copy=copy, track=track)[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/zmq/green/core.py[0m:250: in send_multipart
[1m    msg = super(_Socket, self).send_multipart(*args, **kwargs)[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/zmq/sugar/socket.py[0m:366: in send_multipart
[1m    self.send(msg, SNDMORE|flags, copy=copy, track=track)[0m
[1m[31mvolttron/platform/vip/socket.py[0m:384: in send
[1m    frame, flags=flags, copy=copy, track=track)[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/zmq/green/core.py[0m:215: in send
[1m    self._wait_write()[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/zmq/green/core.py[0m:137: in _wait_write
[1m    self.__writable.get(block=True)[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/gevent/event.py[0m:370: in get
[1m    self._wait_core(timeout, ())[0m
[1m[31m../../../virtualenv/python2.7.11/lib/python2.7/site-packages/gevent/event.py[0m:95: in _wait_core
[1m    result = self.hub.switch()[0m
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <Hub at 0x7fe3a2325910 epoll default pending=0 ref=8 fileno=3>

[1m    def switch(self):[0m
[1m        switch_out = getattr(getcurrent(), 'switch_out', None)[0m
[1m        if switch_out is not None:[0m
[1m            switch_out()[0m
[1m>       return greenlet.switch(self)[0m
[1m[31mE       Failed: Timeout >60s[0m

kmonson, 2016-10-28:
Do we know if this only affects the encrypted tests?

jhaack, 2016-10-28:
Yes, from what I see.

craig8, 2017-01-03:
I believe these have been taken care.

closed 2017-01-03 by craig8