VOLTTRON/volttron #994: ChargePoint API Driver

ffreckle, 2016-10-27:
Kisensum is developing a driver interface to the ChargePoint Web Services API.  It will support the retrieval of basic charging telemetry and the management of charging power limits.  It will be based on the 5.0 version of the ChargePoint API.  The 4.1 version is described [here] (https://na.chargepoint.com/UI/downloads/en/ChargePoint_Web_Services_API_Guide_Ver4.1_Rev4.pdf).

We will publish a design document by November 9 and plan to make a first release available by the end of November.

ffreckle, 2016-11-02:
@jhaack, @craig8, @kmonson 

Would you expect to see one driver agent instance (unique VIP_IDENTITY) for each EVSE ? It seems like this is required to make the pubsub topics work correctly.  To further complicate things, each EVSE can have multiple ports, each with its own telemetry and control.  For maximum flexibility, should we consider one agent instance per EVSE+port ?

ffreckle, 2016-11-02:
An alternative could be to embed the station and port in the point name:

      station01/port02/power

but I haven't experimented and don't know if this would cause other issues.  One agent (process) per station or station+port seems a bit heavy to me but I will follow convention/best practice.

kmonson, 2016-11-03:
Logically splitting up the physical device using different topics is the way we would suggest you handle this. It makes the resulting Agent responsible for a single port and much easier to write/maintain/understand.

To clarify: In the driver framework in VOLTTRON all drivers live under the umbrella of the master driver agent (One VIP_IDENTITY) and the master driver handles the publishing of data to unique topics for each logical device.  If you haven't already you should look at the driver development overview here: http://volttron.readthedocs.io/en/develop/devguides/agent_development/Develop-Driver-Agent.html and the more complete driver interface API docs here: http://volttron.readthedocs.io/en/develop/apidocs/services/MasterDriverAgent/master_driver.interfaces.html

In this case you would point multiple instances of the device driver at the same device with each driver setup for a different port. This is the way we handle BACnet gateways that mirror 100 devices as though it's one 1000 datapoint mega-device.

ffreckle, 2016-11-03:
Thanks Kyle.  I think I've got it.  I will follow up with a sample set of config commands to confirm.

ffreckle, 2016-11-07:
@jhaack 

We will publish the Chargepoint Driver specification on Wednesday.  The plan is to create a new rst file in docs/source/specifications on the develop branch and issue a pull request.  Let me know if we should do something else.

acedrew, 2017-06-06:
This is merged, can we close it @craig8

PR #1074 (2016-12-09): Chargepoint API Driver Version 1.0
The Chargepoint API Driver for VOLTTRON enables real-time monitoring
and control of Chargepoint EVSEs within the VOLTTRON platform
by creating a standard VOLTTRON device driver on top of the
Chargepoint Web Services API.  Each managed chargestation
looks like a standard VOLTTRON device, monitored and controlled
through the VOLTTRON device driver interface. Resolves #994.

---
This change is [[image]](https://reviewable.io/reviews/volttron/volttron/1074)
closed 2017-09-14 by kmonson