VOLTTRON/volttron #995: Should we be stripping off the analysis topic root from the topic string?

craig8, 2016-10-27:
I get why we should do it from devices as the tree is structured campus/building/device/subdevice etc always and forever.

With analysis it's a little bit different/more freeform.  Internally we use a structure such as analysisagentname/campus/building/device/subdevice or whatever amalgamation makes sense.  However because of that the only way to distinguish between an analysis and a device is to look at the analysisagentnames.

So this is the question, should we be stripping off the roots of the tree when entering in the database other than the devices root?

I invite all comments to this

kmonson, 2016-10-27:
Yes we should strip off the analysis root topic. That information is encapsulated in the "source" value in the record that is handed to the concrete historian.

jhaack, 2016-10-27:
But "source" isn't captured anywhere else and there's no way to ask "Give me all analysis results" without explicitly naming your topics analysis/myanalysis/etc. This goes way back to wanting analysis to show up alongside the devices they were being run against:

campus/building/device/point1
campus/building/device/myagent/results1

This may still be a valid use case but right now it's hard to get at all analysis data in a generic way. Maybe we need a source column or put it in its own table like record. That way it's explicitly separated but could still be displayed alongside device data if needed.

schandrika, 2016-10-27:
#750 is the related issue that talks about storing different topics data in different table.

kmonson, 2016-10-27:
@jhaack I would consider that a problem with the concrete historian and the query interface to historians. 

It was an oversight to not provide a way to get at the analytics data through the query interface. We could add a source keyword argument to the query RPC call that takes either a single source or a list and gives you back a single record like it does now or a list of records, one for each source.

schandrika, 2016-10-27:
@kmonson when someone queries by source, say source="analysisagentname" do you mean the historian should return data from topic  analysis/analysisagentname/<topic_name> ? If so, I am little concerned that the topic format "analysis/analysisagentname/<topic_name>" is simply a convention and not something that is enforced.

kmonson, 2016-10-27:
I think he meant analysis/agentname/building...

jhaack, 2016-10-27:
Right, currently it can only be done by convention. If you want to look up your analysis data you need to use something you can identify it with.

schandrika, 2020-02-26:
@jhaack could you comment. I think this issue can be closed since without the root of the analysis topic we can't query it

jhaack, 2020-02-26:
If we could query by source we could get the functionality. Talk to @rlutes if this is still needed.

rlutes, 2020-02-27:
The ability to query by source would be nice but not critical for our current analysis visualizations as we embed the application name in the topic.

jhaack, 2020-02-27:
Thanks, I recommend closing until this becomes a priority

closed 2020-02-27 by schandrika